e2002/yoradio #130: Is it possible to listen to this station?

RingoK, 2024-12-29:
Station here:

https://le05.euddn.net/fe4df3b5d9b59056acb075d82cbaea1f77748a7a562ae518c41b8c95258cdc376e214ca90ae7e16d68e3e80db569184cea5c1203a123ece1502c17fea682a5fdf90d9a5dd449bed4b920defabe9054c46ccbd78abe2ed28575e4125f51563f670ae716199bd82b9c9ad68d9216eb7618005001b1d4ee4913981cf765f79a561511140ec9a954375be50f658a78eedd883229da44ad300ebecf0c7ed82835efd1e5ca33d6eaccfbd786ce068202bab69c/myhits.aac

trip5, 2024-12-29:
I think the address is just toooooooooooo long.  The playlist editor cuts off at 140 characters like so: `https://le05.euddn.net/fe4df3b5d9b59056acb075d82cbaea1f77748a7a562ae518c41b8c95258cdc376e214ca90ae7e16d68e3e80db569184cea5c1203a123ece1502c1`

Where did you get the link?  There must be a shorter URL than that.

RingoK, 2024-12-29:
Shorter url:
https://router.euddn.net/8103046e16b71d15d692b57c187875c7/myhits.aac
This URL crash ESP

Claudiu1975, 2025-01-04:
Is playing just fine on mine.

RingoK, 2025-01-05:
> Is playing just fine on mine.

with i2s dac or vs1053?

Claudiu1975, 2025-01-05:
Sincerely I don't remember if I used vs1053 or CJMCU-1334.