e2002/yoradio #134: ESP32-2432S028- problem with boot

miniarci, 2025-01-13:
I just try to start play with yoradio on ESP32-2432S028 but after upload firmware screen is still blank, no wifi AP and in serial monitor i've bootloop:
 `
ELF file SHA256: c94ca020b

E (475) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
E (483) esp_core_dump_elf: Elf write init failed!
E (487) esp_core_dump_common: Core dump write failed with error=-1
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8
E (455) ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver

abort() was called at PC 0x40121597 on core 0

Backtrace: 0x400828e1:0x3ffe3b30 0x40090129:0x3ffe3b50 0x400965d5:0x3ffe3b70 0x40121597:0x3ffe3bf0 0x40113baa:0x3ffe3c10 0x40082caa:0x3ffe3c40 0x40079772:0x3ffe3c90 |<-CORRUPTED
`
I try on two type options but still the same reaction. Latest myoptions

#ifndef myoptions_h
#define myoptions_h

#define DSP_MODEL DSP_ILI9341
#define DSP_HSPI true
#define LED_INVERT true
#define TFT_CS 15
#define TFT_DC 2
#define TFT_RST -1
#define LED_BUILTIN 17
#define BRIGHTNESS_PIN 21
#define I2S_INTERNAL true
#define PLAYER_FORCE_MONO true
//#define I2S_DOUT 22
//#define I2S_BCLK 16
//#define I2S_LRC 4
#define TS_MODEL TS_MODEL_XPT2046
#define TS_SPIPINS 25, 39, 32 /* SCK, MISO, MOSI */
#define TS_CS 33
//#define IR_PIN 27
#define LIGHT_SENSOR 34
#define AUTOBACKLIGHT_MAX 1024
#endif

Where i should search problem?

ovodomina, 2025-01-14:
Same here. With 2 diferents CYD

miniarci, 2025-01-14:
My board is version with 2 usb, CYD2USB

ovodomina, 2025-01-14:
@miniarci i solved the error changing the esp32 board library to 2.0.17

bf104, 2025-01-16:
> [@miniarci](https://github.com/miniarci) i solved the error changing the esp32 board library to 2.0.17

Thanks so much!
After days of tests and unsuccessful attempts with **ESP32-2432S028R** I was thinking of abandoning the project but this solution worked for me!

miniarci, 2025-01-16:
Me also that help. Issue to close

trex2000, 2025-01-20:
For me it doesn't fix the problem.
If I am not mistaken, the esp32 library was changed to 2.0.17 , right ?
Not the Arduino ESP32 Boards, which is currently it 2.0.18-arduino.5 version, and there seems no 2.0.17 version from this.

The same problem I am getting on serial monitor:
`18:58:09.947 -> ELF file SHA256: 5f02c2e10
18:58:09.947 -> 
18:58:09.947 -> E (483) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
18:58:09.994 -> E (492) esp_core_dump_elf: Elf write init failed!
18:58:09.994 -> E (496) esp_core_dump_common: Core dump write failed with error=-1
18:58:09.994 -> Rebooting...
18:58:09.994 -> ets Jul 29 2019 12:21:46
18:58:09.994 -> 
18:58:09.994 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:58:09.994 -> configsip: 0, SPIWP:0xee
18:58:09.994 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:58:09.994 -> mode:DIO, clock div:1
18:58:09.994 -> load:0x3fff0030,len:4916
18:58:09.994 -> load:0x40078000,len:16436
18:58:09.994 -> load:0x40080400,len:4
18:58:09.994 -> ho 8 tail 4 room 4
18:58:09.994 -> load:0x40080404,len:3524
18:58:09.994 -> entry 0x400805b8
18:58:10.462 -> E (464) ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver
18:58:10.462 -> 
18:58:10.462 -> abort() was called at PC 0x40123d8f on core 0
18:58:10.462 -> 
18:58:10.462 -> 
18:58:10.462 -> Backtrace: 0x400828e1:0x3ffe3b30 0x40090129:0x3ffe3b50 0x400965d5:0x3ffe3b70 0x40123d8f:0x3ffe3bf0 0x401160e2:0x3ffe3c10 0x40082caa:0x3ffe3c40 0x40079772:0x3ffe3c90 |<-CORRUPTED
18:58:10.462 -> 
18:58:10.462 ->`

MaciejMilinkiewicz, 2025-01-26:
Comment those lines:
```c++
#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024
```

trex2000, 2025-01-26:
Hello,
Thanks for the reply.
Why would commenting out these 2 help with the boot issue?
According to this, on pin 34 there is a light sensor and I intended to use it.
The purpose of the next one is not entirely clear, but why should this help solve the partition issue?

MaciejMilinkiewicz, 2025-01-26:
```
ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver
``` 
it means there is probably a problem with the ADC converter code. **LIGHT_SENSOR** uses it.

PhantomTronics, 2025-03-02:
Ok, i'm new with this board, also with the yoradio project...i can't even load the project to the board, i guess it should be loaded to via IDE and the USB connector. IDE gives me all kinds of faults...so i would like to know what sketch do i use to get this board working ?? Can anybody give me a push in the right direction ?