e2002/yoradio #138: ESP32-2432S024 Display issue

philiptbuckley, 2025-01-20:
Has anyone had any success getting YoRadio working with this CYD lookalike?

I set in myoptinos.h:

```
#define DSP_MODEL         DSP_ILI9341
#define DSP_HSPI          true
#define TFT_DC            2
#define TFT_CS            15
#define BRIGHTNESS_PIN    27
#define TSSPI             14, 12, 13    /* SCK, MISO, MOSI */
#define TS_MODEL          TS_MODEL_XPT2046
#define TOUCH_CS          33
```
The display initialises with off-centre logo, but then corrupts the lower (roughly) quarter of the screen after displaying "Trying to <SSID>" and the .... stop progressing across the screen.

It does appear to be successfully processing the stream on the serial console and responding to commands but the web server is unresponsive. Is this a conflict with pins dedicated to WiFi?

[image]

dabu667, 2025-01-20:
Does your CYD board have 2 USB ports (USB-C and microUSB)? Then try these options:
```
/*  DISPLAY  */
#define DSP_MODEL         DSP_ST7789 		 //DSP_ILI9341
#define TFT_CS            15                 /*  SPI CS pin  */
#define TFT_RST           -1                 /*  SPI RST pin.  set to -1 and connect to Esp EN pin */
#define TFT_DC             2                 /*  SPI DC/RS pin  */
#define DSP_HSPI          true               /*  Use HSPI for display  */

/*  TOUCHSCREEN  */
#define TS_MODEL           TS_MODEL_XPT2046  /*  See description/available values in yoRadio/src/core/options.h  */
#define TS_SPIPINS         25, 39, 32    /* SCK, MISO, MOSI */ //dabu667
#define TS_CS              33           /*  Touch screen CS pin  */

#define BRIGHTNESS_PIN     21                /*  Pin for adjusting the brightness of the display (output 0 - 3v3) */
```

It uses a different display driver than the original CYD board.

philiptbuckley, 2025-01-20:
No it just has a single USB-C connector. The board is labelled USB32-024 on the PCB, but I believe it's actually ESP32-2432S024R (the version with the resistive touch). It definitely has an ILI9341 and not an ST7789 according to the schematics which are available [here](https://github.com/NoosaHydro/2.4inch_ESP32-2432S024/tree/main/Downloads). I ordered the board from [Aliexpress](https://www.aliexpress.com/item/1005008205080191.html?spm=a2g0o.order_list.order_list_main.35.657e1802OE4zEd) and despite the diagram and description indicating that it is the twin USB connector CYD it is not. Obviously, I can return it -- but I thought I'd have a play with it first to see if I can make it go. I should have said -- it comes with demo software pre-loaded that demonstrates that the display works perfectly.

dabu667, 2025-01-20:
I have both boards: one with ILI9341 and the second with ST7789 display driver. From what I remember this is the way the display looked when the settings were for the wrong driver. My advice is to give it a go with the settings I sent you - it won't hurt the board and maybe I am right.

IU4ALH, 2025-01-20:
Is not an ili9341   but is a ST7789   driver display !!!

philiptbuckley, 2025-01-20:
Great news! I'll give that a try tonight

dabu667, 2025-01-20:
Try ST7789 driver but ingore my SPI pinout - it is for the 2.8 inch CYD board.

Ladislav217, 2025-01-24:
Hi, I had the same problem. I solved it like this. Enter the change in the sub. C:\Users\snapi\Documents\Arduino\libraries\Adafruit_ILI9341     
Open Adafruit_ILI9341.cpp and edit according

sendCommand(ILI9341_MADCTL, &m, **1 to 0**); //

save and estart arduino

`/**************************************************************************/
/*!
    @brief   Set origin of (0,0) and orientation of TFT display
    @param   m  The index for rotation, from 0-3 inclusive
*/
/**************************************************************************/
void Adafruit_ILI9341::setRotation(uint8_t m) {
  rotation = m % 4; // can't be higher than 3
  switch (rotation) {
  case 0:
    m = (MADCTL_MX | MADCTL_BGR);
    _width = ILI9341_TFTWIDTH;
    _height = ILI9341_TFTHEIGHT;
    break;
  case 1:
    m = (MADCTL_MV | MADCTL_BGR);
    _width = ILI9341_TFTHEIGHT;
    _height = ILI9341_TFTWIDTH;
    break;
  case 2:
    m = (MADCTL_MY | MADCTL_BGR);
    _width = ILI9341_TFTWIDTH;
    _height = ILI9341_TFTHEIGHT;
    break;
  case 3:
    m = (MADCTL_MX | MADCTL_MY | MADCTL_MV | MADCTL_BGR);
    _width = ILI9341_TFTHEIGHT;
    _height = ILI9341_TFTWIDTH;
    break;
  }

  sendCommand(ILI9341_MADCTL, &m, 0); //
}`

philiptbuckley, 2025-01-25:
Perfect @Ladislav217 that's fixed the display issue completely.

Unfortunately, it now reboots with every web config request -- think I may have a memory issue to chase down.

Ladislav217, 2025-01-25:
My conf:

```
#define DSP_MODEL           DSP_ILI9341
#define DSP_HSPI            true
#define LED_INVERT          true
//#define TFT_ROTATE        0
#define TFT_CS              15
#define TFT_DC              2
#define TFT_RST             -1
//#define LED_BUILTIN         17
#define BRIGHTNESS_PIN      27
#define I2S_INTERNAL            true
#define PLAYER_FORCE_MONO           true
#define ENC_BTNR			21
#define ENC_BTNL			22
#define ENC_BTNB			35
#define ENC_INTERNALPULLUP			false
//#define TS_MODEL            TS_MODEL_XPT2046
//#define TS_HSPI            true
//#define TS_SPIPINS          25, 39, 32    /* SCK, MISO, MOSI */
//#define TS_CS               33
//#define SDC_CS 5
//#define IR_PIN              27
//#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024
```

philiptbuckley, 2025-01-25:
My conf was pretty similar to yours -- TS configured as you have commented out -- but I've changed to be identical to yours (plus the portrait/landscape switch in Adafruit_ILI9341.cpp) and it still crashes just as soon as I connect to the yoRadio access point webserver to attempt config. It feels so close! ... so a few more things to check. I have the following setup:

- Arduino IDE 2.3.4
- Espressif ESP32 3.1.1 board support package. Some earlier issues suggest reverting to Arduino ESP32 2.0.18
- _ESP32 Dev Module_ selected in the board manager, same for you?
- I am using partition scheme -Minimal SPIFFS 1.9MB app with OTA 190k SPIFFS- -- same?

After fiddling with other partition schemes I am wondering if I have (partially?) corrupted SPIFFS. I tried _Erase all Flash before sketch upload_, but that made no difference. I've tried reloading using LittleFS data uploader tool -- but it still reboots the board the moment my browser hits 192.168.4.1 so maybe there is a conflict with WiFi and one of the GPIOs.

Ladislav217, 2025-01-25:
I am using ARDUINO 1.8.18, with version 2... I have bad experiences.
I have SPIFFS set to minimum, ESP 32 - 3.1.0

philiptbuckley, 2025-01-26:
Doh! More careful reading of the [Quick Start](https://github.com/e2002/yoradio?tab=readme-ov-file#quick-start) and I might have noticed IDE v2 not supported. 
It was a nightmare installing IDE V1.8.19 tho' as I hit the _esptool not found_ issue, caused by not having `esptool.py` installed. There are lots of notes on how to resolve this, but none worked for me, until I manually installed the tools from the [github repo](https://github.com/espressif/arduino-esp32/tree/master/tools). After running the get.py script it started working and I now have a landscape display, playing through the internal DAC and a speaker connected to the JST connector. Will connect an external DAC next and then try and get the touchscreen working.
What a great piece of code and amazing that it runs on a sub $5 Aliexpress board with integrated display.

RecepAGulec, 2025-01-31:
By the way, a small contribution:
My CYD has 2 USB ports (one micro usb, the other USB c) and the display driver is ILI9341.
The display works normally but the touch does not work.

xek, 2025-02-09:
Thanks for sharing!

I got the touch screen running by setting the same TS_SPIPINS as for the display and modifying `yoRadio/src/core/touchscreen.cpp`:

```
#define TS_SPIPINS          14, 12, 13    /* SCK, MISO, MOSI */
```

```diff

diff --git a/yoRadio/src/core/touchscreen.cpp b/yoRadio/src/core/touchscreen.cpp
index 17d9835..101cd4c 100644
--- a/yoRadio/src/core/touchscreen.cpp
+++ b/yoRadio/src/core/touchscreen.cpp
@@ -49,7 +49,8 @@ void TouchScreen::init(){
       ts.begin();
     #endif
   #endif
-  ts.setRotation(config.store.fliptouch?3:1);
+  ts.setRotation(config.store.fliptouch?0:2);
+  //ts.setRotation(config.store.fliptouch?3:1);
 #endif
 #if TS_MODEL==TS_MODEL_GT911
   ts.begin();
@@ -86,7 +87,8 @@ tsDirection_e TouchScreen::_tsDirection(uint16_t x, uint16_t y) {

 void TouchScreen::flip(){
 #if TS_MODEL==TS_MODEL_XPT2046
-  ts.setRotation(config.store.fliptouch?3:1);
+  //ts.setRotation(config.store.fliptouch?3:1);
+  ts.setRotation(config.store.fliptouch?0:2);
 #endif
 #if TS_MODEL==TS_MODEL_GT911
   ts.setRotation(config.store.fliptouch?0:2);
```

Essentially having the same rotation config as the one set for GT911.

gmx168, 2025-12-31:
Ad CYD 24

if you experience swapped direction in touch interface, better swap here:

 } else { /*     SWIPE TOUCH     */
      direct = _tsDirection(touchX, touchY);
      switch (direct) {
        case TSD_UP:
        case TSD_DOWN: {
       // case TSD_LEFT:
       // case TSD_RIGHT: {
            touchLongPress=millis();

and
  case TSD_LEFT:
       case TSD_RIGHT: {
       // case TSD_UP:
       // case TSD_DOWN: {
            touchLongPress=millis();
            if(display.mode()==PLAYER || display.mode()==STATIONS){
              int16_t yDelta = map(abs(touchStation - touchY), 0, _height, 0, TS_STEPS);
              display.putRequest(NEWMODE, STATIONS);
              if (yDelta>1) {
                controlsEvent((touchStation - touchY)<0);
                touchStation = touchY;
              }

works for any driver.