e2002/yoradio #145: ESP32S3 cannot use VS1053! !

buch1234, 2025-02-05:
ESP32S3 cannot use VS1053! !
VS-MOSI is fixed to pin 23. But ESP32S3 does not have pin 23!
**Fixed miso19 and SCK18 can be found on ESP32S3. Why is there no pin 23?**
How can I connect it?
Can the author release a new version of the program for ESP32S3? Thank you very much!

https://trip5.github.io/yoradio-docs/docs/ESP32-S3-DevKitC-1_44Pin/ESP32-S3-DevKitC-1_44Pin-Board.jpg

e2002, 2025-02-05:
On ESP32S3
The SPI2 (VSPI) default pins are:

gpio.12 SCK Defined as SPI0_SCK
gpio.11 MOSI Defined as SPI0_MOSI
gpio.13 MISO Defined as SPI0_MISO
http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference

closed 2025-02-05 by e2002

buch1234, 2025-02-07:
Thank you very much for your quick reply. But I have been trying for two days and still can't succeed. 

gpio.12 SCK Defined as SPI0_SCK
gpio.11 MOSI Defined as SPI0_MOSI
gpio.13 MISO Defined as SPI0_MISO
I tried multiple connections but failed.
There are also different Rest pin positions.  
I tried many different connections with no success.

Can you make a version for eps32s3?
Thank you very much.

buch1234, 2025-02-10:
After testing over the weekend, it finally worked. It handles the default ISP interface as the author said. Please don't forget to connect REST. Many thanks to the author for his help.
gpio.12 SCK Defined as SPI0_SCK
gpio.11 MOSI Defined as SPI0_MOSI
gpio.13 MISO Defined as SPI0_MISO
don't forget to connect REST！！！