e2002/yoradio #150: ESP32 1732S019 - the display does not work

Ladislav217, 2025-02-26:
Hi all,

I haven't been able to start the display for a week. Could someone help me? Thanks

```
#define DSP_MODEL      DSP_ST7789
#define TFT_WIDTH 170
#define TFT_HEIGHT 320
#define TFT_MOSI            13   
#define TFT_SCLK            12
#define TFT_DC              11
#define TFT_CS              10
#define TFT_RST             1

#endif
```

dabu667, 2025-02-28:
I don't have your board to test it, but I have downloaded the _1.9inch_ESP32-1732S019.zip_ from Ali link you provided and from the schematic it seems the display is using different pins from the ones you mentioned.
[image]
Maybe try these?

nvalw, 2025-03-06:
This module works fine in Yoradio.
Just need to change a few things. You can see it in the photo.
[image]
[image]
[image]
[image]
[image]
[image]

Ladislav217, 2025-03-07:
I don't know what I'm doing wrong but this is the result

[image]

nvalw, 2025-03-07:
[yoRadio-1732.zip](https://github.com/user-attachments/files/19122478/yoRadio-1732.zip)
Special for you... All needed files,

Ladislav217, 2025-03-07:
Thank you very much, everything works as it should.