e2002/yoradio #156: Request for help with correct pins

Poszukiwany, 2025-03-04:
I wanted to make my mother a radio for the kitchen, but since I'm a complete novice, I need a little help.
The board I have is YD-ESP32-S3 N16R8, unfortunately the myoptions.h script does not correctly handle the pins for this board and other similar ones.
I will list here the pins to which it seems to me that a given module should be connected, if I am wrong somewhere, please correct me.

display pins (ILI9341) without touch
MISO-12 MOSI-11 SCK-13 DC-? CS-?
audio (PCM5102A)
Din-4 BCLK-5 LRC-6
sd card mounted on the display (I don't know if it works)
MISO-37 MOSI-35 SCK-36 CS-?
buttons pins
PREV-? NEXT-?
mute pin-21 
wake pin-10
There will be one or two radio stations on the list, so no more buttons are needed.
Currently I have a complicated radio with only pcm5102 on pins 12,13,14 to check if I soldered everything correctly and if the boards work.
I would like an answer for an idiot (take this, put it here)

dorinik, 2025-03-04:
Hi Poszukiwany,
I have the following settings in myoptions.h and it works fine:

#ifndef myoptions_h
#define myoptions_h

#define CLOCKFONT_MONO    false
#define L10N_LANGUAGE     EN

#define DSP_MODEL    DSP_ILI9341
#define TFT_CS          10
#define TFT_RST         -1
#define TFT_DC           9
#define TFT_MOSI        11
#define TFT_SCLK        12
#define BRIGHTNESS_PIN  14

#define I2S_DOUT      16
#define I2S_BCLK      17
#define I2S_LRC       18
#define VS1053_CS     255

#define BTN_CENTER    15    //  CONFIRM  Play, Stop, Show playlist
#define BTN_UP         6    //  Prev, Move Up
#define BTN_DOWN       5    //  Next, Move Down
#define BTN_MODE       7    //  BACK  Mode, Web Radio/SDcard
#define BTN_INTERNALPULLUP    false    //  Enable the weak pull up resistors. Default "true"

#define SDC_CS            39
#define SD_SPIPINS  41, 42, 40        // SCK, MISO, MOSI

#define IR_PIN                1
#define IR_TIMEOUT            80

#endif

Poszukiwany, 2025-03-05:
Thank you for your help and good luck in developing the project

closed 2025-03-05 by Poszukiwany