e2002/yoradio #157: Add Support for Second I2S Stream

Badger101, 2025-03-04:
Hi,

Would it be possible to add the option to have two I2S streams from the ESP32 with two EQ settings (I think the ESP32 has two I2C blocks)?

My use case for this would be to add a sub-woofer speaker (2.1), but you could also make a good bi-wired setup.

dabu667, 2025-03-04:
You don't need two digital i2s interfeces for a 2.1 setup. I used a pcm5102 dac with 3 pam8403 amp boards with proper RC filters on input for each audio canal. It's not audiophile grade but adjusting the gain and having the right speakers i managed to get quite a good sound.

dabu667, 2025-03-05:
You can also use MAX98357A boards and configure them for 2 channels - see this: https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/pinouts#sd-slash-mode-2693328 and then add a crossover filter for 2.1 channels.
Or add a 3rd MAX98357A board and configure it as mix and use it for subwoofer channel with a low pass filter. Anyway, this solution is more expensive.

Badger101, 2025-03-05:
Yes, this is exactly what I'm trying to achieve.

I have two setup in stereo already (great little I.C.'s), and was looking to make a subwoofer to go with it using either the same MAX98357A with the gain upped, or there is a more powerful variant:
https://www.analog.com/en/products/max98365.html

The issue is that these I.C.'s need to 'see' the speaker as it is used for it's own filtering of the PWM output so you cannot use any other filtering (low pass, etc)

'A filterless spread-spectrum modulation scheme eliminates the need for output filtering found in traditional Class D devices and reduces the component count of the solution.' 

https://www.analog.com/media/en/technical-documentation/data-sheets/max98357a-max98357b.pdf

For these the filtering needs to be done in hardware to the I2S stream so I was hoping it would be possible to utilise the second I2S channel for this purpose.

dabu667, 2025-03-05:
I know it's a class D amp and the high frequencies are filtered out by the speaker, but I don't see any reason why a passive filtering (a crossover) won't work on the output of this amp. You can search if indeed they are incompatible, or maybe someone here with a more solid background in audio electronics will help you.

Badger101, 2025-03-05:
That'll be me :)
These types of I.C's (MAX98357A, etc) which don't have any output filtering expect to 'see' (or go into) an inductive load (speaker/headphones).
There not designed to do anything else except to convert the I2S signal into a PWM to go straight to a speaker - this is why is needs to be done in the I2S stream.
Thanks for your help in trying to find a solution!