e2002/yoradio #160: Scrilling text leaves the screen dirty

aFunkyBass, 2025-03-21:
Hello,

I have a 2.4 inches TFT display and I noticed that the when the text scrolls, it leaves some pixels on the screen. I attach a small video to let you see this behaviour.

https://github.com/user-attachments/assets/b9aec7ad-24e9-4dfd-978e-974e05fe6a7f

After the screen have been messed up it stays like that until a full display refresh is done i.e. you use the encoder to change the volume or pick another station.

Has anybody ever experienced something like that?

aly-fly, 2025-04-06:
I think the problem is some Unicode character(s) or something unexpected at the end of the string that messes up the text on the display. String should be cleaned before sending to the display.
Try to dump hex values of the text and see what you get.

aFunkyBass, 2025-04-06:
Hello! I guess it was that radio stations sending strange characters, other stations don't mess the display. Thank you for your help.

closed 2025-04-06 by aFunkyBass