e2002/yoradio #162: 12 hr clock

surfpgh6, 2025-03-27:
Is there any way to change the time to 12hr not 24hr?

aly-fly, 2025-04-06:
Why complicate? Most of the world uses 24hr format ;)

surfpgh6, 2025-04-06:
Only if you are military in the USA.

And I am NOT military!

And if you have ever written C+ code you know its not complicated. With the libraries in an off call scenario I dont know which one the time format is in. So I can change it myself !

aly-fly, 2025-04-06:
Most of the world outside of USA :)  We all know USA has messed-up measurement units ;)
FYI, to change such a "small" thing in a project, you have to make it selectable between both variants, change the online configurator tool, change the configuration webpage on the device, adapt the complete http backend, allocate space on the screen, choose the font and location for am/pm letters, save this setting to flash, and so on. Not simple to do it properly. And test everything.

As you are in an one-off scenario - this is the beauty of open-source software like here, you can fork and adopt the system as it fits to you.