e2002/yoradio #165: Choppy playback, dropouts and crackling noise

aFunkyBass, 2025-04-06:
Hello there I've finally managed to complete my YoRadio project but I'm having a couple of issues. 

1 - The main problem is that I have a lot of dropouts on almost every station I try to listen to. This results in playback being constantly choppy and unpleasant. My Internet connection is quite good, I have plenty of bandwith and very short ping time, nonetheless it seems that the ESP32 can't keep up with the stations. I tried to enlarge the audio buffer setting this `#define AUDIOBUFFER_MULTIPLIER2 12` but I had no luck. I had the sensation that the issue is not caused by network problems, but rather by not enough processing power from the ESP. As a matter of fact everytime I modifiy the volume with the encoder, it seems to lag and then start back playing.
What can I try? Has anybody ever experienced things like that?

2 - I have crackling noise from the speakers that sounds like a ground loop or a RF interference coming from the microcontroller. The project is made up of a ESP32 Dev Kit like this one -> https://www.az-delivery.de/it/products/esp32-developmentboard?variant=36542176914 connected to a PCM5102 audio adapter, a TPA3110 amp, a ILI9341 2.4" display and 2 encoders.  The power source is a 12v wall adapter that directly feeds the TPA3110. The 3.3v complonent are powered by means of a XL4015 buck converter also fed by the same wall adapter. I also have added a 1000 microfarad capacitor at the output of the buck converter. 
How can I eliminate those noises?

Thank you in advance

JakesESP32, 2025-04-08:
Referring the 2nd point, I had the same experience, audio sounding like a ground loop. The only way I could solve it is to use 2 wall adapters, and 2 buck converters, feeding the Esp32 circuit, and the audio amp circuit separately. Not very efficient I admit.

dabu667, 2025-04-08:
1. Could be anything: bad module, bad pcb antenna - check the buffer status, maybe it gives you a hint or an unlucky combo between esp core version and libs version which can overload the CPU.
2. I've built about 6 versions, each with different esp module brand, PCM5102 or max98357 boards, never had a similar problem, but I have to mention that some time the PWM signal from the backlight of the display, if not routed properly may interfere with audio signal - check that - route it away from the amplifier input / shield the amp input wires.

Hope It helps.

aFunkyBass, 2025-04-15:
Hello everyone, to my surprise, issue 1 was due to weak WiFi signal coming to the ESP. I didn't take this into consideration because the access point was quite close to the radio, but there was probably a misalignment or some kind of shield that prevented the wifi signal to reach the receiver. Moving the access point by 20 cm. completely solved my problem... WTF!

I'll work on issue 2 following @dabu667 suggestions. 

Thank you.

JakesESP32, 2025-04-15:
> Hello everyone, to my surprise, issue 1 was due to weak WiFi signal coming to the ESP. I didn't take this into consideration because the access point was quite close to the radio, but there was [...]

1. Delighted to hear that. 

2. I have been experimenting with 600:600 audio transformers between the DAC output and the amplifier input using a single power supply, supplying the 2 buck converters. There was a significant improvement, but still not perfect. Next step I am considering is a ferrite core over the audio cable just before it enters the amp. Shall report once I tried it.

JakesESP32, 2025-04-17:
Ferrite core experiment feedback, no significant improvement.

aFunkyBass, 2025-04-18:
> Ferrite core experiment feedback, no significant improvement.

Too bad...  Thank you for posting your feedback

Lukesan70, 2025-04-27:
For all of the groundloops, forget ferrit cores, adding capacitors, coils etc.
Working with 2 power supplies is also not needed at all.
Using transformers works, but you are actually altering the audio. In- and output impedance, phase etc ... just don't

For all my projects that are subjective to groundloops I use a dc/dc isolator which takes care of these things.  Have a look at for example B0505s
If you google that number you will automatically get result and where to get it.
The 0505 indicates in- and output voltage of the module, so you also have them in different input and output voltages.   You will only need a capacitor like 1 to 10uF at the in- and output. 
You need to take into account is that it also rated for a certain power. So you have them in 1W, 2W etc 

I also use this module in car BT things or those other audio modules. It completely gets rid of the beeps and other digital stuff you hear. A bit magical turning your volume all the way up and hearing no noise after you use it

Of course you still need to keep your cables nice, clean, short etc to minimize the issue of bad signal etc.

JakesESP32, 2025-04-27:
Thank you this looks promising, however, for a 20 + 20W stereo amplifier the power supply calls for a 120W power supply. 

At a 12V supply voltage, the current drawn will be 10A !!!

These devices are rated at 2W .....!!!???

Or do you use this devices on the digital circuit supply only, where you have 5V with a minimum current drawn ...???

Looking forward to your response.

Kind regards

Lukesan70, 2025-04-27:
No worries. No it is not that you have to include the amplifier.  But that was a word of warning since I have seen such a thing that people used that isolator with the amplifier or other things and said that it didn't work. 

Just try on the digital circuit. It also depends, I play a lot with these things.  You might only have to isolate the ESP32 or the PCM5102 or both.  It's just 4 wires in- and output, so a quick test will learn quickly.

Another word of warning ... don't blow up your speakers or get scared... a bit funny but last time I used it to demonstrate at like midnight. It was indeed silent till I pressed play at full volume! :-)

JakesESP32, 2025-04-27:
Thank you for your response .....!!! 💯 

The digital side is the "noise generator" therefore it makes sense to isolate there.

I shall provide feedback, once I tried it ...  👍

aFunkyBass, 2025-04-27:
Hello @Lukesan70 thank you for your advice. I'll get my hands on one of this and try.

JakesESP32, 2025-04-30:
@Lukesan70, Sir you are a genius !!!!. 

I obtained a B0505S-2WR2. The web radio is playing at the moment WITHOUT the annoying high pitch noise or the use of any audio transformers.

The setup is 2 buck converters one at 12V for the amp, and another set at 5V feeding the B0505S-2WR2 and then connected to the digital boards. Both converters are fed from a well regulated bench power supply set at 15V, just because, no particular reason.

I chose to use the B0505S-2WR2 since the data sheet shows a max of 400mA at 5V, I decided I want to have some headroom.

The B0505S-2WR2 circuit was used exactly as the datasheet recommends, with one difference, the datasheet calls for a 4.7uF at the input, I used a 6.8uF because I had it available.

So, the bottom line, this WORKS !!! and the audio quality is AWESOME. !!!

Once again thank you I certainly learned some thing new thanks to you !!!! 👍 

My kind regards from South Africa.

Lukesan70, 2025-04-30:
@JakesESP32     Dis 'n plesier.   Sharing knowledge is caring and especially if it is easy it is most rewarding.

I just helped someone who had the same issue on another forum with a BT call dongle thing in the car.  He bought it like that from the manufacturer who asked a lot of money for a product that sounded bad. I wonder if they actually test their own products lately. 
Since no space inside that dongle I just told him to cut an extension USB cable in half, use that module and use some shrinktube. Issue resolved for like R100.

For the capacitors, if I am in a crazy mood I will place a 100nF capacitor in parallel with the other ones to surpress the spikes even more but that is something which I inherited from the knowledge of my late father. 
There is one thing that you need to know and that is that the output of these modules are not that super well regulated, so the output voltage can depend on the load sometimes but certainly within limits. 

I think you could have actually used a B1205s-2wr2 and eliminated the 5V buck converter, would make it more efficient in times of load shedding :-) but the 0505's are much more common and you make less mistakes using those.

aFunkyBass, 2025-04-30:
Hey thank you for your feedback. 

Like I said in my first post, my project is currently powered by a 12v wall adapter + a buck converter. The wall adapter directly feeds the amplifier and the rest of modules are fed by the buck converter providing 3.3v. The pseudo schematic is represented by the following picture

[image]

Now my idea is to put the isolator just before the audio card. I plan to use a B0303 because I'm using 3.3v to power all the digital modules. The new schematic should be like in the followng drawing

[image]

Do you think this would work or you suggest a different solution?

JakesESP32, 2025-05-01:
I see no reason why your idea will not work.

 However may I recommend the following, instead of a 3v3 buck, use a 5V, which you then use to supply both the ESP, and display. 

The ESP does have a 3v3 output that you can use to power the PCM.

You therefore put the B0505 between the buck and the whole logic system, which isolates it from the analog circuit.

Alternatively you can use a B1205s-2wr2, then you do not need to use the buck converter, but I will be hesitant to use it like this since these isolators are not very well regulated, which may cause other issues.