e2002/yoradio #170: 3.5 inch IPS ILI9488 code modifications?

Another-Taipan, 2025-04-25:
I have a ILI9488 3.5 inch display that uses an IPS panel instead of the usual TFT panel.

When I received the display module from the supplier, it included a note that said "cannot use TN code - contact me for IPS code, otherwise it don't have correct color".

The supplier sent me this text file:

> It is also important to invert the colors, but it is simple, anyone can do it.
> writecommand (0x21);
> 
> 
> writecommand(0xC5); 
> writedata(0x00); 
> writedata(0x4D); //  <--- important!!! -VCOM = -0.79688V
> writedata(0x80); 
> 
> writecommand (0x21); //inverse colors for IPS

I don't fully understand where I need to make these changes in the YoRadio code, so I would appreciate it if someone could please give me some guidance?

My display seems to be working fine without those changes applied, but I really don't know what the correct colors should be.

dabu667, 2025-04-26:
This seems like a required display init sequence. The init for your display type is defined in function `void ILI9486_SPI::init(void)` in file yoRadio/src/ILI9488/ILI9486_SPI.cpp. maybe you have a particular flavor that needs a little extra code. Anyway,  you should probably read the datasheet.

Another-Taipan, 2025-04-27:
I added the following code:

```
// added code for IPS panel
	_writeCommand(0xC5); 
	_writeData(0x00); 
	_writeData(0x4D); 	//  <--- important!!! -VCOM = -0.79688V
	_writeData(0x80); 
	_writeCommand(0x21); 	//inverse colors for IPS
// end of added code
```

to ILI9486_SPI.cpp beginning at line 391, recompiled and uploaded the changed code, but the display looks exactly the same as before ...

So I have either added the code in the wrong place or made some other error, or the added code isn't necessary?

I have a copy of the manual/datasheet from the supplier, but it is in Chinese and I haven't been able to transalet it.  Also, the little bit of English in it refers to a "TFT" panel and not an "IPS" panel.

I am not sure what I should do now ...