e2002/yoradio #171: ##AUDIO.INFO#: slow stream, dropouts are possible ...

Another-Taipan, 2025-04-25:
I have built a YoRadio using an ESP32-S3 N8R2.  

I notice that in the boot log that I sometimes see the following message when connecting to some hosts:

> ##AUDIO.INFO#: slow stream, dropouts are possible

This message only seems to be when I connect to a streaming server that is in another country.  
I do get dropouts when the AAC bit rate is around 200 bps, but don't get dropouts when the bit rate is around 100 bps.

I am puzzled as to what causes this message and why AAC at 200 bps has frequent dropouts.

My radio is about 5m from the WiFi AP and the RSSI is -52dBm, so I don't think it is a WiFi issue.  My internet speed is 100 Mbs.

I can provide more information from the boot log, if that will help to diagnose this issue?

Another-Taipan, 2025-04-28:
I have found where that warning message comes from, it is in "Audio.cpp", in this section:

```
// if the buffer is often almost empty issue a warning  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    if(InBuff.bufferFilled() < maxFrameSize && f_stream){
        static uint8_t cnt_slow = 0;
        cnt_slow ++;
        if(f_tmr_1s) {
            if(cnt_slow > 25 && audio_info) audio_info("slow stream, dropouts are possible");
            f_tmr_1s = false;
            cnt_slow = 0;
        }
```
It looks like the buffer size needs to be increased?

Since I am using an ESP32-S3 N8R2, then there is 2 MB of PSRAM available.  Can I use that PSRAM to increase the buffer size?

If I can use the extra PSRAM, how do I do that?

dabu667, 2025-04-28:
See this from the autor of i2saudio library:
https://github.com/schreibfaul1/ESP32-audioI2S/issues/582
Meanwhile, I think he improved the library byt yoradio uses on older
version.

Another-Taipan, 2025-04-29:
> I think he improved the library but yoradio uses an older version.

I assume that you have tried the latest version of ESP32-audioI2S and was unable to get it to compile or to work successfully?

Did you compile it in the Arduino IDE or in PlatformIO?

Like you, I find it very frustrating that I cannot listen to some streams because of this "slow stream, dropouts are possible" issue.  I wish I was better at coding so I could fix it myself.

dabu667, 2025-04-29:
> I wish I was better at coding so I could fix it myself 
> 

me too :)
I'm using Arduino IDE. 
About a year ago i tried my best to solve this - but I think by best wasn't enough. I installed linux and tried to recompile ESP-IDF using menuconfig to set some parameters for better throughput on ESP32 like:
```
#
# TCP
#
CONFIG_LWIP_MAX_ACTIVE_TCP=512
CONFIG_LWIP_MAX_LISTENING_TCP=512
# CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION is not set
CONFIG_LWIP_TCP_MAXRTX=12
CONFIG_LWIP_TCP_SYNMAXRTX=12
CONFIG_LWIP_TCP_MSS=1460
CONFIG_LWIP_TCP_TMR_INTERVAL=200
CONFIG_LWIP_TCP_MSL=60000
CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT=20000
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=8192
CONFIG_LWIP_TCP_WND_DEFAULT=32768
CONFIG_LWIP_TCP_RECVMBOX_SIZE=32
CONFIG_LWIP_TCP_QUEUE_OOSEQ=y
CONFIG_LWIP_TCP_OOSEQ_TIMEOUT=6
CONFIG_LWIP_TCP_OOSEQ_MAX_PBUFS=4
# CONFIG_LWIP_TCP_SACK_OUT is not set
# CONFIG_LWIP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_LWIP_TCP_OVERSIZE_MSS=y
# CONFIG_LWIP_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_LWIP_TCP_OVERSIZE_DISABLE is not set
CONFIG_LWIP_TCP_RTO_TIME=1500
# end of TCP 
```
.. that I found in an example from schreibfaul1: [https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughpu](https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput) - now the link is dead
Anyway, after many days of frustration I gave up.

I can't be 100% sure, but I've noticed that on some development boards the same code behaves better than others in terms of audio streaming. For example, I replaced the ESP32 module on a CYD board with an original Espressif one and now I don't have any dropouts anymore - except when I turn on the microwave :).

Maybe at some point a very good programmer will come along and manage to implement the latest version of I2Saudio in this project.