e2002/yoradio #173: ESP32+PAM5102A Trouble.

layserk1, 2025-05-04:
Praying for help as i am trying to solve it 3 days already.
The issue is that there is no sound when connecting pam5102a to my esp32. 
ESP and PAM5102a are absolutely fine and 100% working as i uploaded karadio32 on this setup and sound went perfectly.
Did everything as said in upload intructions to this project , generated myoptions file putting ticks to l2s external on base ports, display and encoder. Project uploaded via Arduino IDE perfectly , all booted, AP is on, all settings available, but NO SOUND! I've tried changing giops in myoptions generator, tries copying myoptions file from "examples" folder and uncommenting whats necessary and entering correct giops in it , but that didn't work. 

PLEASE TELL ME WHAT AM I DOING WRONG? Need help badly. Thanks in advance.

[image]

[image]

layserk1, 2025-05-05:
> Hi try this, working here.
> 
> // If I2S DAC used: #define I2S_DOUT 26 #define VS1053_CS 255
> 
> // If VS1053 used: //#define I2S_DOUT 255 //#define VS1053_CS 27
> 
> //External I2S DAC
> 
> #define I2S_DOUT 26 /* DIN connection _/ #define I2S_BCLK 27 /_ BCLK Bit clock _/ #define I2S_LRC 25 /_ WSEL Left Right Clock */
> 
> Rgds

I did this when editing myoptions file from "examples" folder and it looks like this.

[image]

When i generate myoptions file via url https://e2002.github.io/docs/myoptions-generator.html then it looks like this.

[image]

Don't have any idea what am i doing wrong((( I even erasing ESP and freshly upload every sketch-nothing helps...(((

JakesESP32, 2025-05-05:
Try this:

/**/ If I2S DAC used:
#define I2S_DOUT 26
#define VS1053_CS 255**

Before you do this:

//External I2S DAC
#define I2S_DOUT 26 /* DIN connection /
#define I2S_BCLK 27 / BCLK Bit clock /
#define I2S_LRC 25 / WSEL Left Right Clock */

That's the only obvious differences I can see; I am assuming you did do the solder bridges on the 5102, since you said it works somewhere else.

layserk1, 2025-05-05:
> Try this:
>
> /**/ If I2S DAC used:
> #define I2S_DOUT 26
> #define VS1053_CS 255**
>
> Before you do this:
>
> //External I2S DAC
> #define I2S_DOUT 26 /* DIN connection /
> #define I2S_BCLK 27 / BCLK Bit clock / [...]

Isn't commenting vs1053 lines not enough ? I thought commenting it makes it disabled

dabu667, 2025-05-05:
It's right there on hardware setup: 

[image]

JakesESP32, 2025-05-05:
#ifndef myoptions_h
#define myoptions_h

/* Generated by myoptions.h generator - v0.8.1 */
/* https://e2002.github.io/docs/myoptions-generator.html?r=71,43&i=15,16,17&v=27,26,25 */
/* Put this file next to yoRadio.ino and restart Arduino IDE. */
/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */

/*  VSPI PINS.
    SCL(SCK, CLK) must be connected to pin 18
    SDA(MOSI, DIN, SDI) must be connected to pin 23  */
/*  HSPI PINS. Set DSP_HSPI to true
    SCL(SCK, CLK) must be connected to pin 14
    SDA(MOSI, DIN, SDI) must be connected to pin 13  */

/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */

#define DSP_MODEL           DSP_ILI9488  //DSP_ST7796 //DSP_ILI9488 //DSP_SSD1306 //DSP_DUMMY//DSP_2004I2C
#define BRIGHTNESS_PIN      2
#define TFT_CS            5                 /*  SPI CS pin  */
#define TFT_RST           -1                /*  SPI RST pin.  set to -1 and connect to Esp EN pin */
#define TFT_DC            4                 /*  SPI DC/RS pin  */
#define DSP_HSPI          false             /*  Use HSPI for display  */

/*  VSPI PINS. SCL(SCK, CLK) must be connected to pin 18
               SDA(MOSI, DIN, SDI) must be connected to pin 23  */
/*  HSPI PINS. SCL(SCK, CLK) must be connected to pin 14
               SDA(MOSI, DIN, SDI) must be connected to pin 13  */

#define TS_MODEL              TS_MODEL_XPT2046  //TS_MODEL_UNDEFINED//TS_MODEL_GT911
#define TS_HSPI   false      // use HSPI for touchscreen (miso=12, mosi=13, clk=14) instead of VSPI (by default)
#define TS_CS      13
//#define SD_HSPI   false        /* use HSPI for SD (miso=12, mosi=13, clk=14) instead of VSPI (by default) */
//#define SDC_CS      33

/******************************************/

// If I2S DAC used:
#define I2S_DOUT      26
#define VS1053_CS     255

// If VS1053 used:
//#define I2S_DOUT      255
//#define VS1053_CS     27

//External I2S DAC

#define I2S_DOUT      26     /*  DIN connection */
#define I2S_BCLK      27     /*  BCLK Bit clock */
#define I2S_LRC       25     /*  WSEL Left Right Clock */

#ifndef EXT_WEATHER
  #define EXT_WEATHER       true   // Extended weather
#endif

#endif

JakesESP32, 2025-05-05:
Isn't commenting vs1053 lines not enough ? I thought commenting it makes it disabled - **NO the value 255 does that job, if you comment that value or any other value does not do its job in the code.**

layserk1, 2025-05-05:
> Isn't commenting vs1053 lines not enough ? I thought commenting it makes it disabled - **NO the value 255 does that job, if you comment that value or any other value does not do its job in the code.**

#ifndef myoptions_h
#define myoptions_h

/* Generated by myoptions.h generator - v0.8.1 */
/* https://e2002.github.io/docs/myoptions-generator.html?r=71,1,2,4,8,12,43,54&i=5,6,15,16,17,28,29,30&v=21,22,27,26,25,36,34,39 */
/* Put this file next to yoRadio.ino and restart Arduino IDE. */

/* !!! Attention !!! */
/* This configuration depends on some additional libraries. Please install them using the library manager: */
/* Adafruit GFX Library, Adafruit SSD1306, OneButton */

/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */

#define DSP_MODEL			DSP_SSD1306
#define ENC_BTNR			36
#define ENC_BTNL			34
#define ENC_BTNB			39
#define ENC_INTERNALPULLUP			false

#endif

This is myoptions code generated via url. Should i insert this block 

[image]
somewhere here?

layserk1, 2025-05-05:
Ok, gonna try in few hours. Thank you, gonna come with feedback

layserk1, 2025-05-05:
[image]
Now myoptions look like this and still no sound....Any ideas?

JakesESP32, 2025-05-05:
Asking the obvious, did you move the connections to reflect this configuration ???

layserk1, 2025-05-05:
1. Yes, sure.  26 27 25 as in code.
2. If you mean this bridge-its not necessary , everything works in karadio32 without bridge, but i can try...

[image]

3.These connections are necessary to make pam5102a work. Tried without it several times on karadio32 and didnt work

[image]

layserk1, 2025-05-05:
> [image]
> 
> This is not KaRadio, bridge the SCK pads, as well as make the bridges as in the pic above.
> 
> Remove the wires on pins 1-4.
> 
> You must be able to measure conduction between SCK and GND pin on the board.

Oh, I see... thanks, gonna test and come with feedback

layserk1, 2025-05-05:
> [image]
> 
> This is not KaRadio, bridge the SCK pads, as well as make the bridges as in the pic above.
> 
> Remove the wires on pins 1-4.
> 
> You must be able to measure conduction between SCK and GND pin on the board.

Did all you said.[image][image][image]
Bridges made, pins connected , NO SOUND.

layserk1, 2025-05-05:
Checked this pam5102a and same with all bridges made on Karadio32 - works perfectly.... don't have the foggiest idea whats wrong... =(

plin2, 2026-01-04:
Documentation request: put the information regarding soldering of the PCM5102 bridges in the wiring section of the project documentation. 

Additional keywords for searchers so they find this issue:  I2S PCM5102A DAC Decoder GY-PCM5102 I2S Player Module