e2002/yoradio #199: v0.9.555 random kernel panic

jmper-ha, 2025-07-30:
The same HW on 434 works without any problems

Here the log:
```
13:04:30.853 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
13:07:55.400 -> 
13:07:55.400 -> Core  0 register dump:
13:07:55.400 -> PC      : 0x400d4836  PS      : 0x00060830  A0      : 0x800d5584  A1      : 0x3ffb4ef0  
13:07:55.400 -> A2      : 0x3ffc5064  A3      : 0x3ffdff0c  A4      : 0x3ffde628  A5      : 0x00000000  
13:07:55.433 -> A6      : 0x00000000  A7      : 0x000000c7  A8      : 0xbaad5678  A9      : 0x3ffb4ed0  
13:07:55.433 -> A10     : 0x3ffdff0c  A11     : 0x00000008  A12     : 0x3ffdfeec  A13     : 0x00000000  
13:07:55.433 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
13:07:55.433 -> EXCVADDR: 0xbaad5684  LBEG    : 0x40085d51  LEND    : 0x40085d59  LCOUNT  : 0x00000027  
13:07:55.466 -> 
13:07:55.466 -> 
13:07:55.466 -> Backtrace: 0x400d4833:0x3ffb4ef0 0x400d5581:0x3ffb4f20 0x400d55b3:0x3ffb4f40 0x400d55c9:0x3ffb4f60 0x400d55d9:0x3ffb4f80 0x400f87f5:0x3ffb4fa0 0x400f9414:0x3ffb5050 0x400f71b8:0x3ffb5070 0x40093c3e:0x3ffb5090
13:07:55.466 -> 
13:07:55.466 -> 
13:07:55.466 -> 
13:07:55.466 -> 
13:07:55.466 -> ELF file SHA256: a18159d206dc7955
13:07:55.466 -> 
13:07:55.532 -> Rebooting...
13:07:55.532 -> ets Jul 29 2019 12:21:46
13:07:55.532 -> 
13:07:55.532 -> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
13:07:55.532 -> configsip: 0, SPIWP:0xee
13:07:55.532 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:07:55.532 -> mode:DIO, clock div:1
13:07:55.532 -> load:0x3fff0030,len:4832
13:07:55.532 -> load:0x40078000,len:16460
13:07:55.565 -> load:0x40080400,len:4
13:07:55.565 -> load:0x40080404,len:3504
13:07:55.565 -> entry 0x400805cc
13:07:56.558 -> ##[BOOT]#	************************************************
13:07:56.558 -> ##[BOOT]#	*               ёPadio v0.9.555                *
13:07:56.558 -> ##[BOOT]#	************************************************
13:07:56.558 -> ##[BOOT]#	------------------------------------------------
13:07:56.558 -> ##[BOOT]#	arduino:	10819
13:07:56.558 -> ##[BOOT]#	compiler:	12.2.0
13:07:56.558 -> ##[BOOT]#	esp32core:	3.0.7
13:07:56.558 -> ##[BOOT]#	chip:		model: ESP32-D0WD-V3 | rev: 301 | id: 791392 | cores: 2 | psram: 4194304
13:07:56.592 -> ##[BOOT]#	display:	9
13:07:56.592 -> ##[BOOT]#	audio:		I2S (27, 26, 25)
13:07:56.592 -> ##[BOOT]#	audioinfo:	true
13:07:56.592 -> ##[BOOT]#	smartstart:	1
13:07:56.592 -> ##[BOOT]#	vumeter:	true
13:07:56.592 -> ##[BOOT]#	softapdelay:	0
13:07:56.592 -> ##[BOOT]#	flipscreen:	true
13:07:56.592 -> ##[BOOT]#	invertdisplay:	false
13:07:56.592 -> ##[BOOT]#	showweather:	false
13:07:56.592 -> ##[BOOT]#	buttons:	left=255, center=255, right=255, up=255, down=255, mode=255, pullup=true
13:07:56.625 -> ##[BOOT]#	encoders:	l1=34, b1=39, r1=36, pullup=false, l2=255, b2=255, r2=255, pullup=true
13:07:56.625 -> ##[BOOT]#	ir:		255
13:07:56.625 -> ##[BOOT]#	------------------------------------------------
13:07:56.625 -> ##[BOOT]#	CONFIG_VERSION	5
13:07:56.625 -> ##[BOOT]#	SPIFFS mounted
13:07:56.691 -> ##[BOOT]#	display.init	done
13:07:57.156 -> ##[BOOT]#	player.init	done
13:07:57.189 -> ##[BOOT]#	network.begin
13:07:57.257 -> ##[BOOT]#	Attempt to connect to my_home_ap
13:07:57.257 -> ##[BOOT]#	..
13:07:57.887 -> ##[BOOT]#	done
13:07:57.887 -> ##[BOOT]#	netserver.begin	done
13:07:57.887 -> ##[BOOT]#	telnet.begin	done
13:07:57.887 -> ##[BOOT]#
13:07:57.887 -> ##[BOOT]#	Ready! Go to http:/192.168.1.163/ to configure
13:07:57.887 -> ##[BOOT]#	------------------------------------------------
13:07:57.887 -> ##[BOOT]#
13:07:57.955 -> ##CLI.META#: [готов]
[... 116 lines cut ...]
13:34:03.631 -> ##CLI.INFO#
13:34:03.631 -> ##SYS.DATE#: 2025-07-30T13:34:01+03:00
13:34:03.631 -> ##CLI.NAMESET#: 8 TrancePulse
13:34:03.631 -> ##CLI.META#: [соединение]
13:34:03.631 -> ##CLI.VOL#: 39
13:34:03.631 -> ##CLI.PLAYING#
13:34:03.763 -> ##AUDIO.INFO#: format is aac
13:34:03.763 -> ##AUDIO.BITRATE#: 128000
13:34:03.763 -> ##AUDIO.INFO#: icy-name: TrancePulse Dublin
13:34:03.763 -> ##AUDIO.INFO#: AACDecoder has been initialized, free Heap: 120176 bytes
13:34:03.963 -> ##AUDIO.INFO#: stream ready
13:34:03.963 -> ##AUDIO.INFO#: buffer filled in 178 ms
13:34:03.963 -> ##AUDIO.INFO#: syncword found at pos 94
13:34:03.963 -> ##AUDIO.INFO#: syncword found at pos 0
13:34:03.963 -> ##AUDIO.INFO#: Channels: 2
13:34:03.963 -> ##AUDIO.INFO#: SampleRate: 44100
13:34:03.963 -> ##AUDIO.INFO#: BitsPerSample: 16
13:34:03.963 -> ##AUDIO.INFO#: BitRate: 126098
13:34:03.963 -> ##AUDIO.BITRATE#: 126098
13:34:03.996 -> ##AUDIO.INFO#: AAC HeaderFormat: ADTS
13:34:03.996 -> ##AUDIO.INFO#: AAC Codec: MPEG-4
13:34:03.996 -> ##AUDIO.INFO#: AAC Profile: LowComplexity
13:34:03.996 -> ##AUDIO.INFO#: VBR recognized, audioFileDuration is estimated
13:34:04.161 -> ##AUDIO.INFO#: StreamTitle='Nitromethane - Time To Die (Original Mix)'
```

e2002, 2025-07-30:
Ok, I figured out where it is. I need to figure it out.

e2002, 2025-07-31:
I was unable to reproduce the error on a similar hardware configuration during overnight testing.
Will this new 3 files fix the error for you?

[gme_fix_20250731.zip](https://github.com/user-attachments/files/21523928/gme_fix_20250731.zip)

jmper-ha, 2025-07-31:
> What does it mean?

I mean web interface. settings.html

jmper-ha, 2025-07-31:
> check again if the error has disappeared?

Around 3 hour test. Works stable, no errors.

But I found a bug in web editor. Do I have open a new issue or it possible continue in this topic?

e2002, 2025-07-31:
Possible continue in this topic

jmper-ha, 2025-07-31:
For better explanation I made [screencast](https://disk.yandex.ru/i/u9p3cI8du7NNiw)

jmper-ha, 2025-07-31:
First I started twice "live" station but it gives "**connection timeout"**. Next I started "dead" station. it gave **error loading stream**
Then I started the "live" station again, it started with a delay, but showed that the "dead" station stream is playing.

e2002, 2025-08-01:
Here is the corrected `script.js.gz`
Don't forget to clear your browser cache.

[script.js.gz](https://github.com/user-attachments/files/21543009/script.js.gz)

jmper-ha, 2025-08-01:
Hello! Now Playlist editor shows correct stream url. But some streams are not working. For example:

```
http://choco.hostingradio.ru:10010/fm
ESP         -  Ok
PLE         -  Ok
Browser     -  Ok

http://server.audiopedia.su:8000/music128
ESP         -  Ok
PLE         -  Not
Browser     -  Ok

http://s1.voscast.com:10413/stream
ESP         -  Not
PLE         -  Not
Browser     -  Ok
```
Where ESP - HW radio, PLE - PlayList Editor, Browser - just browser (chrome)