e2002/yoradio #213: SPDIF output.

romekb, 2025-09-15:
Hi.
I've created a modified version of the code that supports SPDIF output (COAX/OPTICAL). 
It's available at https://github.com/romekb/yoRadio_spdif
Perhaps you could add this functionality to the main version of eRadio ?
Perhaps you can find a better way to implement this...

MrSzuwarowy, 2026-01-28:
Romek, są gotowe moduły I2S to SPDIF. Sprzętowo łatwiej to zaimplementować, do każdej wersji softu będzie pasować.

GHVSCAR, 2026-10-13:
Dear Romek, I built the Yoradio_spdif version and it works fine but i am having some trouble with the IR. The codes are 100% recognised by the learn mode in the webui and stored also, but the software interprets these 2 'places' to the left. So 3 becomes 1  9=7 etc. Can you help with this?

romekb, 2026-10-13:
1. Upload web files included in my spdif project to radio (from folder data/www)
2. Clear browser cache. 
3. Re-learn remote codes. You should see "Mute" key on remote learn page. If not, clear browser cache or use another browser.
After this keys should work propertly.

GHVSCAR, 2026-10-13:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Thank you! That worked fine.

BartekS1, 2026-10-24:
Hi, romekb

Connecting just a 470-ohm resistor can introduce a DC component with an amplitude of 3.3V.
This may cause strange, unpleasant-sounding crackling noises in the speakers.

You should use a resistive voltage divider to match the signal to the S/PDIF standard (0.5–0.6V) and also filter out the DC component using a film capacitor. See the connection diagram below.

[image]